WouterJD/FortiusANT #475: Issue with FortiusANT: Fortius Not Detected After Reboot

SimonFili, 2025-01-11:
Hi All,

Long story short: I have a Tacx Fortius that worked perfectly until January 1, 2025. I was using TTS4, but it stopped working. After some research, I discovered this GitHub project, and I was thrilled to find a way to keep using my old Fortius!

I proceeded to set everything up:

I installed Windows 10 on my old PC in the gym to start fresh.
Downloaded and installed the USB driver from [this GitHub page](https://github.com/switchabl/t19xx_usb).
Installed FortiusANT.
Connected two USB ANT+ dongles.
Installed Rouvy.
On the first try, everything worked perfectly! I managed to achieve a 30-minute ride in Rouvy—great experience.

However, now when I boot up my Windows 10 system and start FortiusANT, the Fortius is not detected. In Device Manager, I noticed that the Fortius appears under Jungo. When it was working before, it was listed under "LibUSB-Win32 USB Devices".

Here’s what I tried:

I deleted the Jungo device (including the drivers) in Device Manager.
Disconnected and reconnected the Fortius USB cable.
After this, the Fortius appears under "LibUSB-Win32 USB Devices" but with the message “No firmware”, and FortiusANT still doesn’t detect it.
Do you have any recommendations on how to resolve this issue?

Thanks again for creating this amazing tool!

WouterJD, 2025-01-11:
I'll look into it shortly Met vriendelijke groetWouter Dubbeldam

SimonFili, 2025-01-11:
I tried the following:

Disable automatic driver updates in Windows:
This prevents Windows from automatically replacing the libusb driver with the Jungo driver. Here's how:

- Open Control Panel > System > Advanced System Settings > Hardware tab.
- Click Device Installation Settings.
- Select No (to prevent Windows from automatically downloading drivers).

Rebooted, now I see the Fortius and the 2 ANT under the same libusb driver.

Will try to use it soon, reboot and re-test.  Will report back.

I don't mind much doing this on this PC, since it's only used for that in the gym.

WouterJD, 2025-01-11:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Hi Simon from Canada!
I bet you are one of the last TTS4 users, most users ran into a non-supported situation.
What headunit do you have? There are different headunits, some without / with changeable / fixed software.

I do not know the cause of the problem, did not encounter this question before.
It seems you made some steps in the right direction.

Curious how things will turn out.

SimonFili, 2025-01-13:
Did my intro! https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-2587947329

My equipment is this:

USB: Fortius USB T1942 from 2007
Motor : Cosmos & Fortius T1941
Controller/PSU : Cosmos & Fortius 300W T1946.50

ANT USB dongle: https://a.co/d/gfXoXqO

Like I said in the intro, the CPU is running at 100% when biking, will try to update the PC and see if it helps the glitching.  Missing RPM from ANT to Rouvy.

Thanks!

WouterJD, 2025-01-14:
Hi @SimonFili welcome. Let's see how to debug this.
I have a T1932 myself and cannot test/reproduce your situation.

The first step is to copy your console screen (black screen) and see what is reported there.

Do you use the windows executable or the python version?
Since you like to _Grab sport stats data, crunch and analyse_ and _On the first try, everything worked perfectly_ I do not doubt we will get this working.

[usbTrainer.py ](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L434) refers to the intialization procedure, you might start there.

But first, show the console output please; first message expected is `Find and initialise USB head unit`

closed 2025-01-31 by WouterJD