WouterJD/FortiusANT #476: Runoff or Start not enabled

Noud02, 2025-01-11:
Hello,
I installed the drivers and opened the FortiusANT.exe.
The Hardware is found, but the Runoff or Start-buttons are not enabled.

Can someone help me out here?

Cheers Noud

[image]

WouterJD, 2025-01-11:
There is no ANT dongle Met vriendelijke groetWouter Dubbeldam

Noud02, 2025-01-11:
Hey Wouter,
Do I need an ANT dongle to do the Runoff and so on?
The T1932 is connected with Bleutooth.

Cheers Noud

Noud02, 2025-01-11:
*USB I mean.
I have had a look at the manual for FortiusANT again. Now I understand, I always need 2 ANT-dongles to use the FortiusANT, am I right?
Cheers Noud

WouterJD, 2025-01-11:
CorrectMet vriendelijke groetWouter Dubbeldam

WouterJD, 2025-01-11:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Hi Noud

Thanks for sponsoring and I hope to hear from you when FortiusAnt works with the CTP of your choice.

WouterJD, 2025-01-12:
If you have started FortiusAnt with -b or -bb, no ANT-dongle should be required.

Noud02, 2025-01-12:
Hey Wouter, let me introduce myself.
I'm Noud, due to a cycling accident I won't be able to cycle in the 'Wild' for the next few weeks, that's why I've decided to buy a tacx. Normally I cycle all year round on my MTB and in the summer also on my gravel bike.

 On the internet I found a used Tacx I-Flow with T1932 Fortius unit.
 When purchasing, I was not aware that Software activation was no longer possible at Tacx. The old owner had the software activated on his PC and everything worked perfectly. 
I'm looking forward to get `moving´ again :D

WouterJD, 2025-01-12:
Hi @Noud02 thanks for introducing, I like to know who I fis uss with and work for.

What country do you live in?

Noud02, 2025-01-12:
I'm Dutch and live in Germany.

Noud02, 2025-01-18:
Hello, everything works now, with the 2 ANT+ adapter.

I have 1 question/Problem. The HRM(from T1932) in the Runoff is visible in the fortniusANT. When I start the FortniusANT without Runoff, so in Start-Mode, the HRM is not transmitted. Do I have to change some settings to get this to work?

Noud

WouterJD, 2025-01-19:
Hi Noud

I usually do not use the Runoff...

The original idea is that a non coded HRM is connected to the tacx and that that HR is then transmitted through ANT to Zwift.

Its also possible that FortiusAnt connects to your coded HRM, in that case *no* HRM is transmitted to Zwift, because Zwift can read the HRM itself.

closed 2025-01-31 by WouterJD