WouterJD/FortiusANT #477: wxPython

Carlin75, 2025-01-19:
Going crazy here. 

Installed os on ssd and sd card. Tried standard os and Legacy versions (multiple times). But all the time same problem occurs. Can't figure how to execute install wxpython.sh always ends with error wxPython 4.1.1 etc... Is not a supported wheel. 

When I run pi from restart I get values in the terminal but no GUI.

Don't know how to proceed...

WouterJD, 2025-01-19:
Quite busy right now did you read manual? wxP is not supported on all platforms. I have had it working on Raspberry Pi though

Carlin75, 2025-01-19:
No worries! Tried different platforms on raspberry os. No one works. Will continue to find a solution.

WouterJD, 2025-01-19:
As you see in constants.py; FortiusAnt can work without the gui.
The GUI is functionally a nicety; in the end it's only the protocol-converter between tacx and zwift.

Try to get it all working without wxPython, so that your trainer works and then proceed.

WouterJD, 2025-01-19:
Check 4_InstallWxPython.sh (I hope you use that script) and check what path is followed.
Evidently wxPython is not available on all platforms (hardware).

WouterJD, 2025-01-31:
Hi @Carlin75 how about this issue; going the LCD-way now?
I think that is an understandable path.

closed 2025-02-19 by WouterJD