WouterJD/FortiusANT #481: Startup problems.

lt-maverick, 2025-03-30:
Hi Wouter, sorry for the problem I'm about to report.
I've been using FortiusAnt with a Tacx Genius trainer for a while now and first of all I thank you so much for the great job you've done.
Everything is ok if I use version 5.2, but trying with v6.4 and v6.8 the program doesn't start. Initially the error message was related to the lack of the Bless module, but even after installing it the problems continue.
Now the error message is the one you see below:

[image]

I think it's related to the Phyton environment, but I have no idea how to fix it.
Can you please help?

Thank you very much in advance, Ilario.

WouterJD, 2025-03-31:
Benvenuto Ilario - Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2025-03-31:
> Everything is ok if I use version 5.2

You already use FortiusAnt since then?

For now I see three ways to go:

1. Ble/bless is introduced after that and yes - even if you do not use -bb, the library must be present.
Since you do not program in python AND you use Windows, **the easiest solution would be to use the windows-executable**, since then the compiled version contains all logic you need (no local python is required).

2. You could also try to install pysetupdi, using "pip install pysetupdi" since probably it's missing in some dependency tree.
 
3. I could go for implementing a checkup that blebless is importable, but that requires some work.

where option 1 is the easiest. See manual 3.3. But you probable already have downloaded all and you can start with "FortiusAnt (exe).bat" in the Startup folder.

Please let me know...