WouterJD/FortiusANT #483: ANT+ Dongle found but the Trainer isn't

Tofuette, 2025-04-29:
Hello, WouterJD

I plan to use a TacX Vortex T2171 and a TacX ANT+ Dongle T2028 (ANT+ Dongle is detected as Dynastream Innovations Dongle). with no Head unit as it was not included to connect to Zwift through FortiusANT.

- [x] - At First I had trouble getting FortiusANT to detect the ANT+ Dongle and after updating the drivers manually from [Wahoo Support](https://support.wahoofitness.com/hc/en-us/articles/360021559679-Installing-ANT-drivers) 

- [ ] After getting the ANT+ detected by FortiusANT, I have the issue of the Trainer TacX Vortex T2171 still not being detected by FortiusANT, 

- I have checked a similar issue #322 , and according to that issue it was solved through by switching between Garmin Dongle and back to the TacX Dongle. Unfortunately I do not have a Garmin Dongle much less another one, What do you suggest?
- something to note is that the TacX Vortex T2171 Trainer has had the _Orange_ light since initial powering.
- how do I fix this issue?

- If more information is needed to help me, please ask me. I am quite new to this software and tacx.

I hope you'll be able to help me resolve this issue, The images attached are for reference: 

FortiusANT Program

[image]

TacX Dongle T2028

[image]

[image]

TacX Vortex T2171

[image]

[image]

WouterJD, 2025-04-30:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
You already told about your configuration, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

The first question I have is how you start FortiusAnt.
Without parameters, FortiusAnt expects a USB-connected trainer; in your case the "-t vortex" command-line parameter is required. It can also be set using the user-interface and stored as JSON settingsfile.

Please let me know.

Tofuette, 2025-04-30:
Thank you for responding, As per my understanding of the user manual I can just use the Windows Executable to run FortiusANT.

How do I set up the "-t vortex" command-line parameter that is required, or would it be better if I ran the Python script and set it up manually

The only connections I have right now is the USB connected ANT+ TacX Dongle and Power Supply for the Trainer, is there anything I'm missing? I got this trainer off of the second-hand market so I'm not really sure.

WouterJD, 2025-04-30:
Using executable is perfectly fine, check the a scripts how to add the command line parameter.
You can also set interactively (settings, wheelicon)

Python not required

Tofuette, 2025-04-30:
I have set the command line parameter to "-t Vortex" in FortiusANT, so to confirm after Pairing with the Trainer  FortiusANT should run and track the progress accordingly correct?

After fixing the command line issue it is now what I assume to be is pairing but it is talking quite a while
[image]

WouterJD, 2025-04-30:
So far so good.

Now the tacx should respond. I have no such device and never seen, so please check manual and try a bit. 

Letmeknow

Tofuette, 2025-04-30:
Update on my issue It seems to be stuck on Pairing as its been a while now, did I miss another step or there's a separate issue occurring?

WouterJD, 2025-04-30:
I would expect the sequence isok, now it's trying. Did you read the manual? Does it provide hints?

Not online, other than phone, now

Tofuette, 2025-04-30:
I'm still not sure as what is causing the long wait for Pairing but it may be due to the orange light indicator of the Trainer itself though I'm not really sure myself, maybe it really is just taking a long while to connect but based on research the orange indicates that the Brake Firmware needs updating but how can I do that when it won't even connect to the dongle itself.

WouterJD, 2025-05-05:
Hmm... looking at usbTrainer,py asof line 1152, it seems no message is received on the vortex channel as if the trainer does not transmit any message.

Look from line 1224 what messages could be expected.

I have no suggestions for you from here at the moment

closed 2025-05-19 by WouterJD